Infineon/TLV493D-A1B6-3DMagnetic-Sensor #2: TLE493D

nikkadim, 2018-01-10:
Should this library work with new TLE493D sensor? In MagneticSensor3D.h you have some notes about sensor configuration, but according spec this sensor have at least two Fast Mode's (7.5 kHz and 5.7 kHz).

PS: Tried to communicated with TLE493D via this library, no luck.

mhollfelder, 2018-01-12:
Currently, the library has been developed for the TLV493D and will only work with this one.
So it is unfortunately not working with the TLE493D right now, sorry!

We are currently working on some other libraries and I will discuss this one as a request.

Best regards,

Manuel

nikkadim, 2018-01-12:
Thank you @mhoechner! TLV494D works just fine with this lib, than you for your work! Can't wait to try TLE493E with a higher sampling rate.

nikkadim, 2018-03-02:
@mhoechner any updates when TLE493 will be available?
Thank you.

nikkadim, 2018-05-08:
@mhollfelder  terrific! Thank you very much!

mhollfelder, 2018-06-01:
I will close this issue, if you need more help, please let us know.

Best regards,

Manuel

closed 2018-06-01 by mhollfelder

faccolo, 2019-01-02:
@mhollfelder 
Hello,
sorry for the question.
I have TLE493D-W2B6-A3
could i use your arduino lib for this device?
best regards,
Francesco