Infineon/TLV493D-A1B6-3DMagnetic-Sensor #3: Sensor Reads only 0xFF

PeterHuewe, 2018-02-17:
Hi,
any ideas why I can't get measurements from the sensor? I always read back 0xFF.
But the acknowledgement seems to work, 
also the sensors is detected at 0x1F when using a simple i2c slave scanner arduino sketch.

I'm using the examples/cartesian_fast example - on an esp8266.
I added some print statements to the BusInterface.cpp - the result looks like this: 

```
---------BEGIN SETUP------
Addr1F
I2C Read:FF
I2C Read:FF
I2C Read:FF
I2C Read:FF
I2C Read:FF
I2C Read:FF
I2C Read:FF
I2C Read:FF
I2C Read:FF
I2C Read:FF
I2C Send:0
I2C Send:18
I2C Send:0
I2C Send:20
I2C Send:0
I2C Send:9B
I2C Send:0
I2C Send:60
Addr1F
I2C Read:FF
I2C Read:FF
I2C Read:FF
I2C Read:FF
I2C Read:FF
I2C Read:FF
I2C Read:FF
I2C Send:0
I2C Send:18
I2C Send:0
I2C Send:20
I2C Send:0
I2C Send:9B
I2C Send:0
I2C Send:60
I2C Send:0
I2C Send:1B
I2C Send:0
I2C Send:E0
---------BEGIN READ LOOP------
Addr1F
I2C Read:FF
I2C Read:FF
I2C Read:FF
I2C Read:FF
I2C Read:FF
I2C Read:FF
I2C Read:FF
-0.10 ; -0.10 ; -0.10

```
Any ideas?

nikkadim, 2018-02-17:
Would use I2C scanner to check sensor availability and address.

PeterHuewe, 2018-02-23:
Hi,
I debugged the issue and found out the root cause.
The address of the sensor is correct and the sensor works properly, the library also works perfectly.

The issue is caused by using the I2C Pins as digitalPins for the reset logic,
as the ESP8266 does not reconfigure the pins for I2C correctly afterwards.

Looking at the source code here on github, 
the current version does not use the I2C Pins as digitalPins anymore
 - but the v1.0.1 Release zip still does use them
https://github.com/Infineon/3D-Magnetic-Sensor-2GO/releases/download/V1.0.1/IFX-3D-Magnetic-Sensor-2GO.zip
--> Cloning the repo or using the source.zip should do the trick :)

closed 2018-02-23 by PeterHuewe