Infineon/TLV493D-A1B6-3DMagnetic-Sensor #4: 14 15

comarius, 2018-03-20:
What are pins 14 and 15 
pinMode(14, OUTPUT);
pinMode(15, OUTPUT);

mhollfelder, 2018-03-20:
Hi comarius,

These are board specific pins which are connected to the on-board LEDs.

If you have a look on the XMC 2Go implementation for Arduino (the 3D Magnetic Sensor 2Go is based on the XMC 2Go), you will see that pin 14 and 15 is connected to the two additional LEDs of the XMC 2Go. 
The pin out diagram is shown in the wiki [here](https://github.com/Infineon/XMC-for-Arduino/wiki/XMC-2Go).

We are currently rewriting the library and cleaning it up to separate it from from any board specific implementation. 
If you have further questions, please let me know.

Best regards,

Manuel

mhollfelder, 2018-06-01:
We removed these lines from the new examples as they were confusing if the sensor is used on an own board or with other boards from us.

The new release will take care of the changes, in the meantime, we will push the code to the master branch.

Best regards,

Manuel

closed 2018-06-01 by mhollfelder