Infineon/TLV493D-A1B6-3DMagnetic-Sensor #5: Addresses seem to be incorrect

sherylll, 2018-04-23:
https://github.com/Infineon/3D-Magnetic-Sensor-2GO/blob/78c8888f5230a3514f187c201d24e76696890a27/src/MagneticSensor3D.h#L22

For me the library only works with 
´TLV493D_ADDRESS1	=	0x5E,´

nikkadim, 2018-04-23:
void Tlv493d::resetSensor(uint8_t adr)
```
	if (adr == TLV493D_ADDRESS1) {
		// if the sensor shall be initialized with i2c address 0x1F
		mInterface.bus->write(0x00);
	} else {
		// if the sensor shall be initialized with address 0x5E
		mInterface.bus->write(0xFF);
	}
```

nikkadim, 2018-04-23:
it depends on the way you initialize the sensor:
[image]

mhollfelder, 2018-04-25:
Dear sherylll,

I have already an assumption and I think that there might be problems with the library.

Could you please share your hardware setup and a minimum sketch which does not work?
Especially, what kind of PCB (w/ routing information) and what software do you execute?
I will take care that we follow up on this.

@nikkadim: Thank you very much for your support and answers here! Please go forward and let us know if we can support you as well :)

Best regards,

Manuel

mhanuel26, 2018-06-08:
Hello,

I have the demo board TLV493D and I can confirm I haven't been able to make this library work. I have run Cartesian low power and fast examples and always receive 0.1 for the three axes after opening COM port.
My hardware works since I run it with DAVE example and Infineon GUI without problems.

I have change the address as @sherylll suggest without success
#define TLV493D_ADDRESS1			0x5E

@mhollfelder Could you please confirm how this should work on the standard demo board? 

Will appreciate your response,

sherylll, 2018-06-08:
@mhanuel26 have you tried to remove 
`#define TLV493D_ADDRESS1 0x5E`

and really change 
```
typedef enum Tlv493d_Address
{
	TLV493D_ADDRESS1	=	0x1F,
	TLV493D_ADDRESS2	=	0x5E
}Tlv493d_Address_t;
```
to 
```
typedef enum Tlv493d_Address
{
	TLV493D_ADDRESS1	=	0x5E
}Tlv493d_Address_t;
```

mhanuel26, 2018-06-09:
Hello @sherylll ,

When you download the release version as suggested in documentation for arduino integration, the files seems to be outdated, only when I clone the repo I get the above typedef enum in Tlv493d.h file.

I will try to replace the arduino library with clone files and let you know the result.

mhanuel26, 2018-06-09:
Got similar behavior,  after making @sherylll suggested change.

I replace the library folder with cloned folder. After opening COM port all values in zero.

I am using the windows market arduino version 1.8.10

Can you let me know which version and steps needed for a working example please.

sherylll, 2018-06-09:
@mhanuel26 have you also updated the example with the cloned version? This minimum setup worked for me
```
#include <Tlv493d.h>
// Tlv493d Opject
Tlv493d Tlv493dMagnetic3DSensor = Tlv493d();

void setup() {
  Serial.begin(9600);
  while(!Serial);
  Tlv493dMagnetic3DSensor.begin();
}

void loop() {
  Tlv493dMagnetic3DSensor.updateData();
  delay(100);

  Serial.print("X = ");
  Serial.print(Tlv493dMagnetic3DSensor.getX());
  Serial.print(" mT; Y = ");
  Serial.print(Tlv493dMagnetic3DSensor.getY());
  Serial.print(" mT; Z = ");
  Serial.print(Tlv493dMagnetic3DSensor.getZ());
  Serial.println(" mT");
   
  delay(500);
}
```

And I was using Arduino 1.8.5.

Unfortunately I do not have the hardware right now, I'll let you know when I have a chance to test it with the newer Aruidno IDE.

mhanuel26, 2018-06-11:
Thank you very much @sherylll for your support,
It was the wrong IC, now it works with TLE493DW2B6 library.

closed 2018-06-11 by sherylll