Infineon/TLV493D-A1B6-3DMagnetic-Sensor #6: Can't flash with Arduino

exp0nge, 2018-05-23:
When I try to flash with Arduino I get this:
`[Error] Infineon.DebuggerExceptions: It seems that JLink software is not installed please download from www.segger.com and install it. You can specify it by setting java property xmcFlasher.JLink.dllPath `

I have tried the GUI and it detects everything normally. I have the board set to XMC1100 XMC2Go, Programmer AVRISP mkll

mhollfelder, 2018-05-23:
Could you please provide us some more background information on your system?

- What OS and which version are you running?
- What Arduino version do you have?
- What version of J-Link do you have?
- Did you install everything in the standard paths?

The problem is related to not finding the DLL for the J-Link debug probe which is needed to execute the flashing. It is a shared library which the XMC Flasher uses for flashing.

Best regards,

Manuel

exp0nge, 2018-05-24:
I reinstalled JLink to `C:` instead of my other location and it flashed.

closed 2018-05-24 by exp0nge