Infineon/TLV493D-A1B6-3DMagnetic-Sensor #7: Readings stay the same

exp0nge, 2018-05-30:
I've tried both cartesian and spherical sketches and the same readings are printed regardless of how far or the orientation of the magnet.

Readings:
`Amount = 0.17 mT; Azimuth angle (rad) = -2.36 ; Polar angle (rad) = -0.62`

`X = -0.10 mT; Y = -0.10 mT; Z = -0.10 mT`

exp0nge, 2018-06-01:
@mhollfelder Here's a screen shot of the windows software evalkit:
[image]

Here's my arduino config:
[image]

mhollfelder, 2018-06-01:
Thank you very much for the screenshot, this already resolved the issue:

The libraries which are currently available for Arduino are only for TLV493D-A1B6 and TLE493D-W2B6 3D magnetic sensor variants.

As shown in the evaluation software, you have the TLE493D-A2B6 which is different from the other two sensors in terms of internal sensor read outs as well as the I2C protocol. Therefore, this library will not work for that sensor which explains why no sensor values are read out.

On which timeline do you need a library?
I will try to trigger the development and align internally for the library for that sensor if needed.

Best regards,

Manuel

exp0nge, 2018-06-01:
@mhollfelder I'm entering this - https://www.hackster.io/contests/Infineon3D

Is it possible to flash what the evalkit windows software uses then read out via serial?

exp0nge, 2018-06-01:
It looks like I can flash with what the Windows software uses but since it starts out in power off and I don't know what to write to the registers to start it, I can't get anything using the serial connection.

If you can tell me what I need to write to what address, I can at least progress somewhere depending on how the actual serial output is being displayed for the windows software. Unfortunately that software doesn't look to be open source.

mhollfelder, 2018-06-03:
You can find the user manual/datasheet of the sensor here:

User Manuel [TLE493D-A2B6](https://www.infineon.com/dgdl/Infineon-Infineon-TLE493D-A2B6-UM-v01_00-UM-v01_00-EN.pdf?fileId=5546d46262b31d2e01633b4bba5f3c5a), the datasheet is stored here [TLE493D-A2B6](https://www.infineon.com/dgdl/Infineon-TLE493D-A1B6-DS-v01_00-EN.pdf?fileId=5546d462602a9dc801606f142b5c7fe5).

You would need to adapt the existing libraries to work with the registers and start the sensor in the respective mode/setup. Please note that the sensor does not support repeated starts in the I2C protocol, i.e. you would need to rewrite parts of the implementation to take care of this. 
Additionally, one would need to check whether the Wire library works with that sensor out of the box.

I will align with my colleagues how to proceed here and come back with a reply.

Best regards,

Manuel

exp0nge, 2018-06-05:
Wow actually https://github.com/Infineon/TLE493D-W2B6-3DMagnetic-Sensor does seem to work, not sure why it didn't before. Thanks for the hints @mhollfelder

closed 2018-06-05 by exp0nge

adammetcalf, 2019-10-28:
@mhollfelder  hi Manuel,

Is there any news on a timescale for the above mentioned Library for the A2B6?

Best wishes,
Adam

mhollfelder, 2019-11-05:
Hi Adam,

we already published a repository for the A2B6 variant as well:

https://github.com/Infineon/TLE493D-3DMagnetic-Sensor

There are some issues but my latest knowledge is that the internal development team is working on fixing it. Basic reading of the sensor should work - this was the latest status which I am aware of.

I hope this gives you a first starting point :)

Best regards,

Manuel