Infineon/TLV493D-A1B6-3DMagnetic-Sensor #9: Adafruit HUZZAH32

tpitman, 2018-11-11:
I am trying to use the TLV493D-A1B6 board with a Adafruit HUZZAH32 Arduino dev  board.

I started with the magnetic sensor board pictured at the top of this github readme file.

I broke away the small sensor board from the main board so I could connect the HUZZAH32 to it. I made a cable that lets me connect it back to your board and that works fine.

If I use the PC software to get values everything works fine. On the scope I see several bytes of clock and data over and over really fast.

I install this github library and it compiles and builds.

I connect it to my HUZZAH32 board and I get nothing.

When I watch the start up on a scope on the i2c lines I see the clock going and data going so I know I have the correct pins.

The problem is that when it tries to read data I get nothing. All of the examples do the same things. They just send an 8 byte worth of clock over and over about every second and nothing else.

During initialization I do see 4 bytes of clock and data, but am not sure what that means.

How do I figure out what is going wrong?

tpitman, 2018-11-11:
More information:

The code here is giving bus errors right at the very start.

I put some serial output and traced that in the tlv493d::writeOut every time it tries to write there is a bus error.

Apparently the interface->bus->endTransmission() is never 0 like it should be.

Any ideas how to fix this?

mhollfelder, 2018-11-11:
Hi tpitman,

Thanks for your message - I will try to give you some answers to your questions:

> Another thing. When I run the PC software the first thing that seems to go out on the i2c bus is FF twice and then 00 twice. While the FF is going out the TLV493D doesn't respond and then while the 00 is going out it does.

If you have a look at the [user manual](https://www.infineon.com/dgdl/Infineon-TLV493D-A1B6_3DMagnetic-UM-v01_03-EN.pdf?fileId=5546d46261d5e6820161e75721903ddd), you will see that this is a reset sequence of the sensor. Just have a look on page 21 and the following ones.

It could be that the HUZZAH32 has some timing issues and the address is not set correctly from the library. 

What you can do is the following and we could try a quick test with two functions:

- Ensure that you have pull-ups added to your board for `SDA/SCL`
- Use the function `Tlv493dMagnetic3DSensor.begin(Wire, TLV493D_ADDRESS1, false);`
- Use the function `Tlv493dMagnetic3DSensor.begin(Wire, TLV493D_ADDRESS2, false);`

Just replace the respective line in one of the examples, I used the cartesian one. Please test it one after another, i.e. first with `TLV493D_ADDRESS1` and in a separate try with `TLV493D_ADDRESS2`.
`false/true` enables the reset of the sensor and tries to set the respective address - it could be that the HUZZAH32 has some issues with timing as the Arduino library writes only single `0xFF` or `0x00` during reset. 

Your finding indicate that there might be a problem when setting the correct addresses.

> Apparently the interface->bus->endTransmission() is never 0 like it should be.

Mainly, one would need to ensure a stable voltage on the `SDA/ADDR` pin when the sensor starts.

However, I did a test with the MH ET LIVE ESP32MiniKit and I had no issues (except some missing `return` statements which were added by 97068923d5c8920f3928124373e797703ff128e8 and ddf73f3956148ec6e23f13721fa8069e8f83e2dc in development branch):

[image]

[image]

tpitman, 2018-11-12:
I tried what you suggested with the begin function and it didn't make any difference.

I also noticed that your changes you mentioned were in the develop branch, so I switch to that branch. That didn't make a difference either.

I am not sure the esp32 library is working right. As indicated before the first thing that always comes out of the esp32 is the address 0. I never see the 0xFF initialization byte that you are talking about.

I notice this 0xFF 0x00 sequence when I watch what the Infineon dev board does, but not when I have it connected to the esp32 adafruit feather board.

It seems like the esp32 board I have is not doing the reset sequence. Could that cause the sensor to never work right? Is there a way to have the sensor just always initialize itself and not need the controller to do it?

I also tried plugging in the sensor after the esp32 board was up and running just to make sure there were voltages on the SCL and SDA lines like you mentioned. Is there something special I need to do there?

This esp32 board is all 3.3v, so that should not be a problem (I know the sensor can't handle 5v.)

What else can I try here?

tpitman, 2018-11-12:
Some testing results.

I now have an analog scope connected instead of a digital one.

With

Tlv493dMagnetic3DSensor.begin(Wire, TLV493D_ADDRESS1, false);

The first thing I see on the scope is: 0xBC and the 9th bit is high meaning the slave didn't reply

With

Tlv493dMagnetic3DSensor.begin(Wire, TLV493D_ADDRESS2, false);

The first thing I see on the scope is: 0x3F and the 9th bit is high meaning the slave didn't reply

With

Tlv493dMagnetic3DSensor.begin(Wire, TLV493D_ADDRESS1, true);

The first thing I see on the scope is: 0x00 and the 9th bit is high meaning the slave didn't reply

With

Tlv493dMagnetic3DSensor.begin(Wire, TLV493D_ADDRESS2, true);

The first thing I see on the scope is: 0x00 and the 9th bit is high meaning the slave didn't reply

So this seems to me that the sensor needs the initialization, but for some reason the Wire library isn't allowing it to go through or your library isn't sending it correctly.

The slave never acks anything.

The scope is showing what seems to be good signals. They look very much like the ones I see when I use the full Infineon dev board that works fine.

So why does the sensor not respond to my esp32 but does respond to the Infineon dev board (and to your esp32 board)? What could possibly be the difference?

tpitman, 2018-11-12:
I also just noticed that when the Infineon dev board starts up the very first thing I see on the scope is the SCL and SDA lines coming up and the very second they are just above 3 volts the clock starts clocking and the SDA is kept high. Basically sending the 0xFF. then there is  about 10us where both SCL and SDA are high and then the FF is sent again. Then 10us and the clock goes for 9 bits while the SDA is low. That also happens twice.

Basically what I uploaded as the second picture above. FF is sent twice and the 00 is sent twice.

On the FF the sensor doesn't ack. on the 00 the sensor does ack. Either that or it is the dev board keeping the SDA low. I don't know.

So is this why the Infineon dev board works and my esp32 doesn't? I don't know how I am supposed to send this FF FF 00 00 sequence from my board as it powers up.

mhollfelder, 2018-11-12:
Thanks for your investigation - this indeed seems to be a strange problem.
Additionally, the sensor seems really not responding to anything.

Could you do me a favor and send the exact name written on the bottom of the board?

Is really TLV493D-A1B6 written on it or maybe TLE494D with an E? 
Would also be great to have a picture of both sides to ensure that you have the right board for this library.

I just would like to avoid that you have another chip version for which we have another library and which is not supported by this library.

Best regards,

Manuel

tpitman, 2018-11-12:
Sorry for all the updates, but I am testing and trying to post as much information as I can.

Since my huzzah32 always seems to send the slave address as the first and only byte, I decided to play around with the Tlv493dMagnetic3DSensor.begin function. I found something interesting.

It would appear that the address I pass into the begin function is left shifted by one bit and the last bit before the slave ack bit is always 1.

Below I have included images from my scope for the following commands:

Tlv493dMagnetic3DSensor.begin(Wire, 0x00, false);
Tlv493dMagnetic3DSensor.begin(Wire, 0x01, false);
Tlv493dMagnetic3DSensor.begin(Wire, 0x02, false);
Tlv493dMagnetic3DSensor.begin(Wire, 0x10, false);

As you can see the lowest bit is always 1 and the other bits are shifted left one bit.

I tested just sending Wire.beginTransmission / Wire.endTransmission with the same numbers and they all came out fine.

Looks like the Tl493D library is doing something weird with the address.

Please explain?

This still doesn't tell me how to get this device to send the wake up sequence to the sensor, however. Is there some way to tell the Wire protocol to not send the device address and send raw data like FF FF 00 00?
[image]
[image]
[image]
[image]

tpitman, 2018-11-12:
The board I am using is the one on the left pictured in the readme for this repo:

https://github.com/Infineon/Assets/blob/master/Pictures/3D%20Magnetic%20Sensor%202Go.jpg

I cut the sensor part away from the main board. I then put a header on both halves so that I could reconnect the sensor to your dev board and continue to test with it. That is what is working when I reference things working.

I then made a cable with embedded resistors in it to connect to the huzzah32.

The green thing is my bracket to mount inside the thing I am building to use the sensor.

As I said above connecting through the header still works fine when connected to your infineon board.

I have 2 cables, one with 10k and one with 1.2k. Doesn't make a difference which I use. You can see the signals in the images above show good square values.

Following are pictures:

[image]
[image]
[image]
[image]

mhollfelder, 2018-11-12:
>Tlv493dMagnetic3DSensor.begin(Wire, 0x00, false);
>Tlv493dMagnetic3DSensor.begin(Wire, 0x01, false);
>Tlv493dMagnetic3DSensor.begin(Wire, 0x02, false);
>Tlv493dMagnetic3DSensor.begin(Wire, 0x10, false);
>
>As you can see the lowest bit is always 1 and the other bits are shifted left one bit.

What you see is the expected behavior as you are giving a 7 Bit address to the I2C master and depending on Read (R) or Write (W) operation, the address is shifted one bit left and inserted. The LSB is the R/W bit and is sett accordingly:

More information here as well: 
https://www.totalphase.com/support/articles/200349176-7-bit-8-bit-and-10-bit-I2C-Slave-Addressing 

Per my understanding, the behavior for ` Wire.beginTransmission / Wire.endTransmission` is strange as I would expect there the bit shift as well. Especially, the library for TLV493D only calls the underlying functions like `Wire.beginTransmission`.

You have the right board and it is all strange why you see the behavior. Especially, as I could test it with a TLV493D sensor and the ESP32 successfully.

Could you please send me a picture of the front as well? 
The pull-ups are not located on the breakable end, are they?

A check from me: 
Have you tried using the XMC-for-Arduino port in combination with the TLV493D library with the original board? Is the board working with XMC and TLV493D library in combination with the Arduino framework not using the GUI?

Thank you very much and best regards,

Manuel

tpitman, 2018-11-12:
Pardon my ignorance with i2c 7 bit addresses. That makes sense.

I will try what you suggest. The break off board does not have the resistors on it. They are on the main xmc board.

I hesitate to try the XMC library and flash the XMC board for fear that if it doesn't work I won't have any way to continue testing the magnetic sensor at all. I still need to continue with the rest of my project and prototyping and can't afford to risk that.

I too am confused why this doesn't work. Maybe it is this specific adafruit esp32 board. Sadly I really need to use it for other reason. Here is a link:

https://www.adafruit.com/product/3619

Can you explain why the XMC board is able to do the FF FF 00 00 sequence and why that doesn't happen on my board?

If the sensor never gets that sequence should it still respond? If not then that is the whole problem. How would we solve that?

Here is the front image. Those wires are for connecting the scope and don't affect the operation.

[image]

mhollfelder, 2018-11-12:
Quick response:

I assume, you can create the same reset behavior by replacing the `resetSensor()` function in `Tlv493d.cpp` with the following function:

```C++
void Tlv493d::resetSensor(uint8_t adr)
{
	mInterface.bus->beginTransmission(0xFF);
	mInterface.bus->write(0xFF);
	mInterface.bus->write(0x00);
	mInterface.bus->write(0x00);
	mInterface.bus->endTransmission(true);
}
```

This does not take care of specific reset behavior or the reset function.
Use this with the normal `begin()` function without the parameters and it should do the job.

I don't get why my ESP32 works and the HUZZAH32 does not work with the TLV493D sensor.
This is really strange from my point of view.

Best regards,

Manuel

tpitman, 2018-11-12:
I take it back. I tried this with another nRF52 adafruit device and this one works. I am getting sensor data from the sensor.

So now the big question is why won't this work with the adafruit huzzah32?

tpitman, 2018-11-12:
I just tried your idea of changing the reset sensor function. That doesn't work because the sensor never acks anything. I don't think it acks the reset sequence, does it?

tpitman, 2018-11-16:
Does this sensor need to be communicated with at a certain speed? How about the signal shape, does it have to be special?

mhollfelder, 2018-11-19:
The signal shape is the normal signal shape of any I2C communication and should have specific rise time and signal shape as usual:

https://hackaday.com/2016/07/19/what-could-go-wrong-i2c-edition/ 

Did you analyze the analogue signal shape as well? 
Taking your twisted cables into account, this could be an issue as well.

For the signal speed: if you have a look in the [datasheet](https://www.infineon.com/dgdl/Infineon-TLV493D-A1B6-DS-v01_00-EN.pdf?fileId=5546d462525dbac40152a6b85c760e80), you see the following speed information on page 11:

[image]

You could try to change the communication speed as well to see whether it makes some changes.
I try to get a HUZZAH32 to do some more debugging for you as well - I think without having the board I cannot test it as you do.

Do you have additional insights right now?

Best regards,

Manuel

markotikvic, 2020-01-30:
I know this issue is closed, but I am facing same issues as @tpitman (the sensor doesn't ACK after address byte is sent) so I was wondering if he ever found out what the problem was?