Infineon/TLV493D-A1B6-3DMagnetic-Sensor #11: STM32

akundigraber, 2019-03-29:
Hello,

our project is based on STM32 Plattform. Is the Tlv493d Library also available for STM32?
We get compiling errors for STM32.  it is working fine with a ATMEGA328 chip.

Regards,
Armin

jaenrig-ifx, 2019-03-29:
Hi Armin,

Only Arduino platforms are supported. If the STM32 platform you are using is Arduino compatible, and the corresponding "Wire.h" adaption for their low level driver peripherals is available, it should at least compile. Otherwise, you are on your own :) and you have to provide this Wire I2C low level drivers adaption for your specific platform and MCU.
Can you share your build error log?

Best Regards,
Juan A

mhollfelder, 2019-03-29:
Hi Armin,

Thank you very much for your report - I just jump in to help out here and support my colleagues.

To proceed we need some more information and details:
Could you provide us some more information and what the exact error is, e.g. is it not compiling at all or do you just see no output? 
Do you use the Arduino platform on the respective MCU or some generic platform?
We would need some more background to help you out here.

Please keep in mind that we only support the Arduino platform in this repository, i.e. if you run Arduino on the respective controller we can help with the Arduino library for the TLV493D, but we cannot give support if the problem comes from the underlying implementation of Arduino for your respective platform. 
Additionally, for other systems, this library would need to be ported and changed accordingly.

Thank you very much and best regards,

Manuel

akundigraber, 2019-03-29:
Hi Manuel,

thank you for the quick reply. Yes we use the Arduino platform on the MCU.
thats the error we get:

Build options changed, rebuilding all
/Users/ak/Documents/Arduino/libraries/TLV493D-A1B6-3DMagnetic-Sensor-master/src/Tlv493d.cpp: In member function 'void Tlv493d::resetSensor(uint8_t)':
/Users/ak/Documents/Arduino/libraries/TLV493D-A1B6-3DMagnetic-Sensor-master/src/Tlv493d.cpp:287:29: error: call of overloaded 'write(int)' is ambiguous
   mInterface.bus->write(0x00);
                             ^
In file included from /Users/ak/Documents/Arduino/libraries/TLV493D-A1B6-3DMagnetic-Sensor-master/src/Tlv493d.h:38:0,
                 from /Users/ak/Documents/Arduino/libraries/TLV493D-A1B6-3DMagnetic-Sensor-master/src/Tlv493d.cpp:34:
/Users/ak/Library/Arduino15/packages/adafruit/hardware/nrf52/0.9.3/libraries/Wire/Wire.h:56:12: note: candidate: virtual size_t TwoWire::write(uint8_t)
     size_t write(uint8_t data);
            ^
In file included from /Users/ak/Library/Arduino15/packages/adafruit/hardware/nrf52/0.9.3/cores/nRF5/Stream.h:26:0,
                 from /Users/ak/Library/Arduino15/packages/adafruit/hardware/nrf52/0.9.3/cores/nRF5/HardwareSerial.h:24,
                 from /Users/ak/Library/Arduino15/packages/adafruit/hardware/nrf52/0.9.3/cores/nRF5/Arduino.h:56,
                 from /Users/ak/Documents/Arduino/libraries/TLV493D-A1B6-3DMagnetic-Sensor-master/src/Tlv493d.h:37,
                 from /Users/ak/Documents/Arduino/libraries/TLV493D-A1B6-3DMagnetic-Sensor-master/src/Tlv493d.cpp:34:
/Users/ak/Library/Arduino15/packages/adafruit/hardware/nrf52/0.9.3/cores/nRF5/Print.h:50:12: note: candidate: size_t Print::write(const char*)
     size_t write(const char *str) {
            ^
exit status 1

mhollfelder, 2019-03-29:
Hi Armin,

I just edited your comment to format it as a code block with console output.
Helps me to digest the information in an easier way ;)

You should be able to resolve the issue by changing the following line `287` in:

`/Users/ak/Documents/Arduino/libraries/TLV493D-A1B6-3DMagnetic-Sensor-master/src/Tlv493d.cpp`

Change it from `mInterface.bus->write(0x00);` to `mInterface.bus->write((uint8_t)0x00);`

The problem is that the write function is overloaded and the compiler cannot resolve the correct function, per my understanding. 
Here, we introduce a typecast `(uint8_t)0x00` to explicitly tell the compiler which function should be called.

I could not test it, but it should work - please let me know whether you face problems.

Best regards,

Manuel

akundigraber, 2019-03-29:
Hi Manuel,

Thank you for the hint! It works now ;)
one quick question, what could be the reason for the instability of the Sensor values?
Amount = 0.17 mT; Azimuth angle (rad) = 0.79 ; Polar angle (rad) = -0.62
Amount = 0.31 mT; Azimuth angle (rad) = 0.32 ; Polar angle (rad) = 0.00
Amount = 0.22 mT; Azimuth angle (rad) = 0.00 ; Polar angle (rad) = -0.46
Amount = 0.10 mT; Azimuth angle (rad) = 1.57 ; Polar angle (rad) = 0.00
Amount = 0.24 mT; Azimuth angle (rad) = 1.11 ; Polar angle (rad) = -0.42
Amount = 0.22 mT; Azimuth angle (rad) = -0.46 ; Polar angle (rad) = 0.00
Amount = 0.33 mT; Azimuth angle (rad) = 0.32 ; Polar angle (rad) = -0.31
Amount = 0.24 mT; Azimuth angle (rad) = 0.46 ; Polar angle (rad) = -0.42
Amount = 0.29 mT; Azimuth angle (rad) = 1.11 ; Polar angle (rad) = -0.73
Amount = 0.17 mT; Azimuth angle (rad) = 0.79 ; Polar angle (rad) = -0.62
Amount = 0.37 mT; Azimuth angle (rad) = 0.59 ; Polar angle (rad) = -0.27
Amount = 0.37 mT; Azimuth angle (rad) = 0.59 ; Polar angle (rad) = -0.27
Amount = 0.33 mT; Azimuth angle (rad) = 0.32 ; Polar angle (rad) = -0.31
Amount = 0.22 mT; Azimuth angle (rad) = 0.46 ; Polar angle (rad) = 0.00
Amount = 0.29 mT; Azimuth angle (rad) = 0.79 ; Polar angle (rad) = -0.34
Amount = 0.17 mT; Azimuth angle (rad) = 0.79 ; Polar angle (rad) = -0.62

The Magnet are not moved between the first and last line of Serial output.

BG,
Armin

tpitman, 2019-03-29:
As a user I can tell you this is just because it is so sensitive. You will
find that the best solution is the average the values.

mhollfelder, 2019-03-29:
Hi Armin,

It's easy to understand if you keep in mind how the principles of error stochastics work and what is the typical error of the sensor.

If you take a look at the [datasheet](https://www.infineon.com/dgdl/Infineon-TLV493D-A1B6-DS-v01_00-EN.pdf?fileId=5546d462525dbac40152a6b85c760e80), you will find on page 17 the resolution (assuming 12 Bit accuracy):

- Resolution 12-bit readout: 0.98 – µT/LSB

The Amount is calculated via:

[image]

As you see from the formula, without going into details, the error of every axis would sum up for the magnitude of the magnetic field, i.e. the 0.1 mT would sum up and the fluctuations can be easily explained.
If you are interested, you would need to search for _propagation of errors_ and you should find some more information - formulas get quite fast quite unhandy ;)

Best regards,

Manuel

bilalahmad333, 2019-05-07:
Hello, 
I have some question in continuation to this thread. I am getting some measurement as well but they are quite random e.g my raw measurement i.e. converting data to signed 12 bit integer values are as follows

 Bx = 0, By = 0  Bz = -1 temp = 342 
  Bx = 0, By = 1  Bz = 0 temp = 341 
 Bx = -2, By = 0  Bz = 1 temp = 342 
 Bx = 0, By = -1  Bz = 1 temp = 342 
 Bx = -1, By = -1  Bz = 0 temp = 342 
  Bx = 0, By = 1  Bz = 0 temp = 341 
 Bx = 1, By = 1  Bz = 1 temp = 343 
 Bx = -1, By = 2  Bz = 0 temp = 341 
 Bx = -1, By = -1  Bz = 0 temp = 342 
 Bx = -2, By = 1  Bz = 0 temp = 341 
 Bx = 2, By = 1  Bz = 0 temp = 342 
 Bx = 0, By = 0  Bz = 1 temp = 342 
 Bx = 0, By = -1  Bz = 0 temp = 341 
 Bx = 0, By = 1  Bz = 0 temp = 342 
 Bx = -1, By = 0  Bz = 0 temp = 342 
  Bx = 2, By = 1  Bz = 0 temp = 342 
  Bx = -2, By = 1  Bz = 0 temp = 341 
Bx = -1, By = 0  Bz = 0 temp = 341 
 Bx = 0, By = 1  Bz = 0 temp = 341 
 Bx = 3, By = 0  Bz = 0 temp = 342 
 Bx = 1, By = 0  Bz = -1 temp = 342 
 Bx = 2, By = -1  Bz = 1 temp = 342 

Now I am not sure whether I am reading the measurement wrong or these are the values expected from sensor. I am not touching the sensor and it is lying on the table. Additionally, I am using Nordic NRF52 controller, sensor  TLV393D-A1B6 and I using the driver given on infineon official website.

svsingh120, 2019-12-17:
Hi, Can u please help me with how to implement implement the code of interface.h in tlv493d infeneon driver for nrf52

OlafFilies, 2019-12-17:
> nrf52
Hello, which sensor board do you have, the one with USB or without USB?

svsingh120, 2019-12-18:
I am having nrf52832 pca10040 dev board. I have tried with driver code of tlv493d defining the read and write command. Is power enable and power disable as well as set_addr_wait is also required. Please help

svsingh120, 2019-12-18:
@OlafFilies  nd thanku for replying. Please help i am stuck into this setup. I don't know what to do further.

OlafFilies, 2019-12-18:
Sorry, but I have to ask again. Do you have an eval board with the mounted Infineon Sensor which looks like the ones in the pictures here [https://github.com/Infineon/TLV493D-A1B6-3DMagnetic-Sensor](url) (left or write one ?) or do you have a bulk chip. This make more of a difference than the micro controller itself.
If I know that, I can give you a specific answer, but anyway, if you want to use this library with a Nordic micro controller, than must at least exchange a all lines of code which use Arduino specific commands for GPIOs, SPI, Serial.print with the appropriate Nordic commands.

svsingh120, 2019-12-18:
No, I am using sensor hardware externally, it not on dev board. As well as I am using nrf52 14.2 sdk implementing i2c. 
Also I have changed functions of this arduino code which is working fine with arduino. 
I have used nrf_drv_twi.c tx and rx fuctions as read and write functions. 

https://arduino.stackexchange.com/questions/53882/communicating-with-magnetic-sensor-tlv493d-a1b6-over-i2c
I have tested it on arduino which gives raw data. 

But it not working with nrf.

svsingh120, 2019-12-19:
Nrf dev board is having USB. I am printing data through NRf_log_info

OlafFilies, 2019-12-19:
Ok, so thats a sensor which is not on a evaluation board, but I still have question about the Sensor itself, as we have two nearly identical Hall Sensors. One is the TLE493D and the other is the TLV493D (E  = automotive versus V = industrial). This little difference makes a huge difference in the interface as both Sensors have a different register setting and there are two different Arduino libs for them. So be sure which one you.
Also we are working on a library setup which will cover different ecosystems like Ardunino, Infineon XMC, raspberryPi and other. This will take some time and it will not directly cover Nordic µC but it will more easy to exchange the controller dependent part from the independent.
As we do not have a Nordic µC board here, we can only give you some hints for conversion to the Nordic ecosystem. There is a PaltformIO port for the Nordic  nRF52, so have a look here [https://platformio.org/platformio-ide?utm_source=github&utm_medium=xmc-for-arduino](PlatformIO with vscode).
If this works, you have a good chance that the Sensor library will also work without changes if not, I can only give you some hints where to look.

- If it is a TLV493D than have a look at Tlv493d.h, there are two addresses defined try them both.
- look at the Tlv493d.cpp begin function, the Sensor is set back to factory defaults at startup
- switch of the lowpower mode
- power enable switches the Sensor on if it is on the eval. board a bulk Sensor should be always switched on as soon as it gets power

svsingh120, 2019-12-19:
Thanks a lot
I will check and reply back soon.

svsingh120, 2019-12-19:
@OlafFilies 
I am working on KIEL. Please tell me which IDEAS would be better for me to install from platform.

svsingh120, 2019-12-19:
I have worked on stm4stm32 which is eclipse based. Will eclipse would be better for me. Also I want to develop the skills in programming. What would be good option for me.

OlafFilies, 2019-12-19:
For PlatformIO you can use Microsofts vscode. Have a look [https://github.com/Infineon/platformio-infineonxmc](here), it explains how to install for the Infineon XMC µC but afterwards you can change the used platform from XMC to Nordic or any other supported platform.

svsingh120, 2019-12-22:
Thanks alot...  @OlafFilies  it's working perfect on platformio. Now I am trying to implement ble peripheral and trying to send data through a beacon

closed 2020-01-14 by jaenrig-ifx