Infineon/TLV493D-A1B6-3DMagnetic-Sensor #12: Cartesian_low_power example always showing same data

mlvandenbos, 2019-04-03:
When trying the library with a '3D magnetic sensor 2GO' board, with example 'Cartasian_low_power' is always shows this in serial monitor:

[image]

I'm using the arduino IDE v1.8.5 and library v1.0.1.
The board itself is working fine in the 3D toGo IDE.

The output of the compiler/uploader:

```
--------------------------
Infineon XMC Flasher Lite
Copyright Infineon Technologies 2017
--------------------------
Windows 10
Operating System: Windows 10

Initialisation
--------------------------
Windows 10
Loading JLink Library... 
JLink Library Version: V6.44d

Configuring Device
Device Name : XMC1100-0064
Interface   : SWD
Speed       : 4000
File Upload
--------------------------
Uploading C:\Users\xxxxxx\AppData\Local\Temp\arduino_build_506221/Cartesian_low_power.ino.hex
... Done
--------------------------
... Finished Succesfully
```

jaenrig-ifx, 2019-04-03:
Hi Mark,

Certainly I reproduce the same error. It works with the shield2go ( + xmc2go) variant. 
Yes, it seems like a bus error. I reopen it here on the right repository. We will work on it and get back to you as soon as possible.

Best Regards,
Juan A

jaenrig-ifx, 2019-04-30:
Dear Mark,

Sorry for the delay. We are now looking into this issue. 
Of course, your insights and findings would be highly appreciated :) 
Thanks for you patience. 

Best Regards,
Juan A

jaenrig-ifx, 2019-05-14:
Dear Mark,

Thanks for you patience. I have created a branch [kit2go-wa](https://github.com/Infineon/TLV493D-A1B6-3DMagnetic-Sensor/tree/kit2go-wa ) with a workaound for the TLE493D-A1B6 Kit2go. The documentation is not updated, and the power pin is controlled by a pin, that needs to be enabled. 
It should work now if you download that branch. 

Best Regards,
Juan A

mlvandenbos, 2019-05-15:
Hi Juan,

It is working with the kit2go-wa branch. I could simply compile and upload the 'Cartesian' example and receive the data.
Thank you.

Mark

closed 2019-05-15 by mlvandenbos

giftofyours, 2022-02-19:
Hi Juan,

I seem to be encountering a similar issue as Mark, getting readings:
X = 0.00 mT; Y = 0.00 mT; Z = 0.00 mT

Meanwhile, the device works smoothly in the GUI application. The GUI application shows that I have device TLV493D-A1B6.
The board itself is labelled "TLV493D-A1B6_2GO_V3.2"

Tried to use the branch https://github.com/Infineon/TLV493D-A1B6-3DMagnetic-Sensor/tree/kit2go-wa that you shared on 14 May 2019, but that branch does not seem to exist any more. 

Would appreciate advice how I can get the code working, or how I can edit the library directly to enable that power pin. 

Thanks!
Gi