Infineon/TLV493D-A1B6-3DMagnetic-Sensor #13: TLV493D-A1B6-3D with nordic Nrf52 unstable values

bilalahmad333, 2019-05-07:
Hello,
 I am getting some measurement but they are quite random e.g my raw measurement i.e. converting data to signed 12 bit integer values are as follows

Bx = 0, By = 0 Bz = -1 temp = 342
Bx = 0, By = 1 Bz = 0 temp = 341
Bx = -2, By = 0 Bz = 1 temp = 342
Bx = 0, By = -1 Bz = 1 temp = 342
Bx = -1, By = -1 Bz = 0 temp = 342
Bx = 0, By = 1 Bz = 0 temp = 341
Bx = 1, By = 1 Bz = 1 temp = 343
Bx = -1, By = 2 Bz = 0 temp = 341
Bx = -1, By = -1 Bz = 0 temp = 342
Bx = -2, By = 1 Bz = 0 temp = 341
Bx = 2, By = 1 Bz = 0 temp = 342
Bx = 0, By = 0 Bz = 1 temp = 342
Bx = 0, By = -1 Bz = 0 temp = 341
Bx = 0, By = 1 Bz = 0 temp = 342
Bx = -1, By = 0 Bz = 0 temp = 342
Bx = 2, By = 1 Bz = 0 temp = 342
Bx = -2, By = 1 Bz = 0 temp = 341
Bx = -1, By = 0 Bz = 0 temp = 341
Bx = 0, By = 1 Bz = 0 temp = 341
Bx = 3, By = 0 Bz = 0 temp = 342
Bx = 1, By = 0 Bz = -1 temp = 342
Bx = 2, By = -1 Bz = 1 temp = 342

Now I am not sure whether I am reading the measurement wrong or these are the values expected from sensor. I am not touching the sensor and it is lying on the table. Additionally, I am using Nordic NRF52 controller, sensor TLV393D-A1B6 and I using the driver given on infineon official website.

jaenrig-ifx, 2019-05-14:
Dear bilalahmad333,

"I using the driver given on infineon official website" means you are using this libs?
https://www.infineon.com/cms/en/product/evaluation-boards/tlv493d-a1b6-ms2go/#!tools

BR,
Juan A

bilalahmad333, 2019-05-14:
Yes, this one

9Volts9er, 2019-05-20:
Hi,

so you want to know if the noise on the value is acceptable for this sensor, right?

Ok, here we're talking about (highest) 3 LSB of noise. With a resolution of 0.098mT/LSB this would be ~0.3mT.
In the datasheet of the sensor, there is specified a value of typically 0.1mT magnetic noise.
Link to the datasheet: [TLV493D - datasheet](https://www.infineon.com/dgdl/Infineon-TLV493D-A1B6-DS-v01_00-EN.pdf?fileId=5546d462525dbac40152a6b85c760e80)
Also there are some other factors that may increase the sensor noise: There's an offset on the values, a sensitivity drift, magnetic hysteresis, etc.
Regarding all this effects, I would say that 3LSB of noise are totally acceptable for an uncalibrated sensor.

If you want to have a more stable value, you can take the average of - let's say - 10 readouts to reduce the noise.
Also you could implement some kind of offset-calibration to reduce your error.
The offset-calibration would be to take the average of e.g. 1000 readouts with no magnetic field applied, and substract this value from every new measurement.
If you want to be extremly precise, you can even apply a fixed known magnetic field to the sensor and calibrate the linearity of the sensor.
But I would say for most applications it's enough to implement a rolling average, often not even that is necessary ;)

I will close the issue for now, if you have any further problems, please feel free to reopen.

closed 2019-05-20 by 9Volts9er

svsingh120, 2019-12-17:
Hello, I am bit mediocre to programming. Can u please give me some support, how you have interfaces the tlv493d driver code from infeneon with nrf52 sdk. I'm using sdk14. 2.
I have tried but i dont know how to write power enable , powerdisable function code for nrf52 for adding in interface.h. 
Please help me with this. I will be gratefull

svsingh120, 2019-12-17:
I have given nrf tx and nrfrx function for read and write operation. But i am getting the garbage data. Help me in procedure please