Infineon/TLV493D-A1B6-3DMagnetic-Sensor #16: TLV439D-A1B6 no I2C communication

BEN-ZZ, 2020-09-22:
Anyone got the examples in the repository to work? 

I am using TLV493D-A1B6 MS2GO and TLV493D-A1B6-3DMagnetic-Sensor-1.0.3 library. The sensor reading is 0 for all the axis. When I used Infineon's GUI (which confirmed the sensor is TLV not TLE), everything is fine, so l know the hardware is fine.

Do l really need to change
  
typedef enum Tlv493d_Address
{
	TLV493D_ADDRESS1	=	0x1F,
	TLV493D_ADDRESS2	=	0x5E
}Tlv493d_Address_t;

to

typedef enum Tlv493d_Address
{
	TLV493D_ADDRESS1	=	0x5E
}Tlv493d_Address_t;

What did your guys change to get the example to work?

I did not have issue with the TLE repository for single 3D sensor. The TLE system does work funny when you use two or more sensors. I was hoping TLV will give me better i2c performance.

Thanks.

Regards

Ben

gwideman, 2020-09-27:
For what it's worth, I am compiled the Cartesian example as-is (for Teensy 3.6), with sensor attached to SCL0/SDA0, and it ran with no changes required. 

I have noticed though that my Tlv493d.h has this:

`
typedef enum Tlv493d_Address
{
	TLV493D_ADDRESS1	=	0x5E,
	TLV493D_ADDRESS2	=	0x1F
}Tlv493d_Address_t;
`
... which indeed has the effect of your proposed change. And that matches the version of Tlv493d.h in the master branch currently. So I'm not sure why your version of that header file has 0x1F as the first address. Did you somehow get an old version of the library from somewhere?

gwideman, 2020-09-27:
Stupid github "insert code" is broken, apparently.
For what it's worth, I am compiled the Cartesian example as-is (for Teensy 3.6), with sensor attached to SCL0/SDA0, and it ran with no changes required. 

I have noticed though that my Tlv493d.h has this:

 typedef enum Tlv493d_Address
 {
     TLV493D_ADDRESS1 = 0x5E,
     TLV493D_ADDRESS2 = 0x1F
 } Tlv493d_Address_t;

... which indeed has the effect of your proposed change. And that matches the version of Tlv493d.h in the master branch currently. So I'm not sure why your version of that header file has 0x1F as the first address. Did you somehow get an old version of the library from somewhere?

BEN-ZZ, 2020-09-29:
Thanks, Gwideman,

I am trying to get both 0x5E and 0x1F work, because l am going to use two 3D hall sensors. I can see the sensor i2c address is hard-coded, which is the same as some people say here. I was in last week. I will look into that in this week.

I am using the latest library. 

Regards

Ben

sfeutrier, 2020-10-01:
I was also stucked with all values to 0. If you look at the .cpp file line 70, you need to define TLV493D_A1B6_KIT2GO macro (or just comment the ifdef) when using the board. 

This fixed my problem

BEN-ZZ, 2020-10-03:
I need to use two TLV, so I used two TLV eval boards,  but jumped wires so that one TLV will get power right after startup, and the other one will get power from the GPIO for LED2. I did comment out the ifdef, but added:

void Tlv493d::begin(int SenNo)
{
  if (SenNo==0)
    begin(Wire, TLV493D_ADDRESS1, true);
  else if (SenNo==1)
    begin(Wire, TLV493D_ADDRESS2, true);
}

void Tlv493d::begin(int SenNo, TwoWire &bus)
{
  if (SenNo==0)	
    begin(bus, TLV493D_ADDRESS1, true);
  else if (SenNo==1)
    begin(bus, TLV493D_ADDRESS2, true);
}

void Tlv493d::begin(TwoWire &bus, Tlv493d_Address_t slaveAddress, bool reset)
{
	/**
	 * Workaround for kit2go vdd enable
	 */
	//#ifdef TLV493D_A1B6_KIT2GO

  if (slaveAddress==0x5E) 
  {
    pinMode(LED2, OUTPUT);	
    digitalWrite(LED2, LOW);
    //delay(50);
  }
  else if (slaveAddress==0x1F)
  {
    digitalWrite(LED1, HIGH);
    pinMode(LED2, OUTPUT);	
    digitalWrite(LED2, HIGH);
    delay(50);
  }

	//#endif

In Arduino program:

    Tlv493dMagnetic3DSensor1.begin(0);
    //Tlv493dMagnetic3DSensor1.setAccessMode(4);
    delay(100);
  
    Tlv493dMagnetic3DSensor2.begin(1);
    //Tlv493dMagnetic3DSensor2.setAccessMode(4);
    delay(100);

As l can see, **the issue with the above approach is the start of the second sensor always kicks out the first one from the I2C network**. From the TLV user manual, 

"Configuration with two slaves
Slave #0 is powered up together with the whole system start up, while slave #1 remains powered down. Within
the first 200 μs after the power-up, slave #0 reads the voltage applied on SDA / ADDR pin. If the voltage level
on ADDR is high, the address is set to “1” (default case with open drain configuration). If the voltage level on
ADDR is low the address is set to “0”. This configuration remains fixed till the next power-down or reset.
Once the 200 μs have transcurred, the master powers up slave #1 with one of the I/O lines. The master drives
the SDA / ADDR pin to the opposite level at which slave #0 was configured. After 200 μs slave #1 is also
configured."

Questions are:
1. Do l need to externally pull-up/down the SDA pin outside of the library? I could not find **the place in the library where SDA is pulled up and pulled down**. I did see both 5E and 1F i2c address works individually, but not both with TLV in a i2c network. 
2. I can see the library uses more than 200us to set up tlv i2c connection. Is the 200us requirement very important?

Maybe some guru or Infineon people can help. TLE and TLV are very good sensor, but it does take time to work with the i2c setup.

Regards

Ben

sfeutrier, 2020-10-05:
I will just put forward a few hypotheses..

I figured out that on the board TLV493D_A1B6_KIT2GO (since v3.2), LED2 macro in arduino is now wired on VDDSens which is the voltage up for the onboard sensor (and also the pull-up). LED1 macro now turns on the real led LED2.
In the Tlv493d::begin(TwoWire &bus, Tlv493d_Address_t slaveAddress, bool reset) function you use LED2 for pull down and pull up but by doing that, you just turn off the sensor (is it what you want ?)
You should have at least one line where you manually pull down or up the SDA macro directly ( digitalWrite(SDA..., HIGH/LOW);

All of this if your board is in version 3.2 (see this link for the mess : https://www.infineon.com/dgdl/Infineon-Infineon-3DMS2GO_TLv493D-A1B6-UM-v01_02-EN-UM-v01-UserManual-v01_02-EN.pdf?fileId=5546d462525dbac40152ac4ca1d318c2)

BEN-ZZ, 2020-10-19:
My board is in version 3.2. I got it run now. so l will close this isue.

closed 2020-10-19 by BEN-ZZ