Infineon/TLV493D-A1B6-3DMagnetic-Sensor #19: Dual sensor example?

aramaerospace, 2020-11-25:
Is it possible to post some simple example for dual TLV493D sensors? I follow the power up procedure described in the user manual, but with no luck.
Thanks!

9Volts9er, 2021-04-09:
Hi, I have created an example here:
[dualSensor.ino](https://github.com/9Volts9er/TLV493D-A1B6-3DMagnetic-Sensor/blob/multipleSensor/examples/DualSensor/DualSensor.ino)

I think, this should work. But unfortunately I don't have two sensors here to test. I will try to get a second one, but in the meantime maybe you or anyone else wants to test it.

If it works I can create a Pull Request to include the example in the original library. Maybe I can also try to create an example for 8 sensors as well (should be possible according to the datasheet), but as I see it this will require some deeper changes in the library.

9Volts9er, 2021-04-13:
I've got access to two sensors recently and managed to debug my sketch. 
With the new changes it works now, at least for my setup. I have a [TLV493D-A1B6 Shield2Go](https://www.infineon.com/cms/en/product/evaluation-boards/s2go_3d-sense_tlv493d/), which is stacked onto a [TLV493D-A1B6 MS2Go Kit](https://www.infineon.com/cms/en/product/evaluation-boards/tlv493d-a1b6-ms2go/).

The working example is included into Pull Request #21.

aramaerospace, 2021-05-05:
This seems to be working. Thanks!

closed 2021-05-05 by aramaerospace