Infineon/TLV493D-A1B6-3DMagnetic-Sensor #20: Sensor2Go not powered

brato, 2021-04-06:
I chose the XMC2Go Board to flash the program. The Sensor does not get 3.3 Volts on the dedicated pin as probably port 15 is not set to HIGH.

https://github.com/Infineon/TLV493D-A1B6-3DMagnetic-Sensor/blob/ea11b706d0f7b0f3bed0c0808f7eae22afb1f7eb/src/Tlv493d.cpp#L70-L74

I think the issue is in the above code.

9Volts9er, 2021-04-07:
That's true.
The problem is that `TLV493D_A1B6_KIT2GO` is not defined anywhere in the example.
A solution would be to copy and paste this code into your `setup()` like this:

```
void setup() {
  Serial.begin(9600);
  while(!Serial);

  //Inserted code
  pinMode(LED2, OUTPUT); 
  digitalWrite(LED2, HIGH); 
  delay(50); 
  
  Tlv493dMagnetic3DSensor.begin();
}
```

Or you simply define `TLV493D_A1B6_KIT2GO` in the "Tlv493d.h".

brato, 2021-04-07:
Thanks for the quick reply . I missed that the library is also for the shield 2 go. Otherwise I would have suggested to remove the ifndef entirely as eg in the [Tle493D library](https://github.com/Infineon/TLE493D-3DMagnetic-Sensor/blob/master/src/Tle493d.cpp) line 36, as it is quite frustrating that the Sensor2go example doesn't work out of the box. What about adding a comment to the example?

9Volts9er, 2021-04-08:
I've added some code lines and comments for this to the examples. Can you have a look to my Pull Request #21 and tell me if this solution is good? Or should I cange anything?

brato, 2021-04-08:
Yes excellent. Thank you very much!

closed 2021-04-08 by brato