Infineon/TLV493D-A1B6-3DMagnetic-Sensor #22: 3+ sensors

electronicDuck, 2022-01-31:
Hi there, thank you for the library as is, works great. Do you have any examples for 3+ sensors, my project uses 3 hopefully but from reading seems like once you go above 2 the process for 3-8 seems to be different to the dual example given? Thank you in advance

9Volts9er, 2022-02-01:
Hi,

so basically the principle for 3+ sensors stays the same as for two:
Power up one sensor at a time and configure it to some not yet used address.

But the problem lies in the detail. So the sketch for 2 sensors works like this:
- When powering up sensor 2, the I2C-interface is not yet configured, so the SDA-pin of the sensor sees a LOW
- By default, the IICAddr bits in the sensors MOD1 register are 0b00
- That means, the sensor 2 will respond to I2C-address 0x1F
- When powering sensor 1, the SDA-line is already driven by the microcontrollers I2C-hardware, so it's pulled HIGH
- Again the IICAddr bits in the sensors MOD1 register are 0b00
- So the sensor 1 will now reply to I2C-adress 0x5E

Here Table 6 from the datasheet, where it shows all different sensor addresses and how the configuration needs to be for them:
[image]

For more than two sensors this principle from above breaks down, as they all get configured to address bits 0b00 by the library in the begin() function.
That means you will need to change the library. For less or equal to 5 sensors this wouldn't be too difficult.
Simply configure the first sensor, where the SDA-line is LOW to address bits 0b00.
Then afterwards SDA will be HIGH, so configure the next sensor and set its address bits to 0b11.
Then the next to 0b10, then 0b01 and the last one can have 0b00 again.

For more than 5 sensors however you need to actively drive the SDA-line when powering up the sensor. This is pretty difficult to include into the library, because the SDA-pin is not defined by default, and it has different pin numbers for different microcontroller boards.
For example the XMC2Go has its SDA on digital pin 11 (Arduino numbering), while for the XMC1100 Bootkit it is digital pin 15.

There is a chapter in the sensors [User Manual](https://www.infineon.com/dgdl/Infineon-TLV493D-A1B6_3DMagnetic-UM-v01_03-EN.pdf?fileId=5546d46261d5e6820161e75721903ddd) describing the configuration steps for multiple sensors in detail.

If there is some demand, maybe we could think of changing the library a little, so that the user is able to define the SDA pin number. Then the library could drive this pin actively when configuring the sensors. 

Or you call `Wire.end()` before powering up the sensors where the SDA-line should be LOW. But I'm not sure, if this command really pulls the pin LOW for any microcontroller...

electronicDuck, 2022-02-02:
Hi 9Volts9er,

Thank you for your swift response and detailed reply. I am familiar with Arduino which is what I am using for this but this will be the first time I am delving into libraries. I'm using 3 TLV493D's with a Sparkfun Pro Micro (Atmega 32u4). 
I'm in at the deep end, can you give me a nudge in the right direction, which parts of the library do I need to be looking at? And then I need to add a part which configures the address, have I got it right I need to write to the device with said address? Read and write or just the IICAddr bits?

Thank you for your help and sorry for the basic questions!

9Volts9er, 2022-02-02:
Hi electronicDuck,

well, basically your main changes can be made in the [Tlv493d.cpp](https://github.com/Infineon/TLV493D-A1B6-3DMagnetic-Sensor/blob/master/src/Tlv493d.cpp). There in the [begin()](https://github.com/Infineon/TLV493D-A1B6-3DMagnetic-Sensor/blob/8945ae676571233a9a974c6e8b0b053e1faf52d7/src/Tlv493d.cpp#L65) you find the following lines:

```
// get all register data from sensor
tlv493d::readOut(&mInterface);
// copy factory settings to write registers
setRegBits(tlv493d::W_RES1, getRegBits(tlv493d::R_RES1));
setRegBits(tlv493d::W_RES2, getRegBits(tlv493d::R_RES2));
setRegBits(tlv493d::W_RES3, getRegBits(tlv493d::R_RES3));
// enable parity detection
setRegBits(tlv493d::W_PARITY_EN, 1);
// config sensor to lowpower mode
// also contains parity calculation and writeout to sensor
setAccessMode(TLV493D_DEFAULTMODE);
```
With the first line all sensor registers are read from the sensor into variables on the microcontroller.
This variables can then be manipulated to match your desired configuration and then they will be written back to the sensor.
With `setRegBits()` you can make modifications to single bits in the variables. For example the `setRegBits(tlv493d::W_PARITY_EN,1);` sets the parity-enable-bit to `1`.
In that manner you can also set your `IICAddr` Bits. You can find the names of the bits in the [Tlv493d_conf.h](https://github.com/Infineon/TLV493D-A1B6-3DMagnetic-Sensor/blob/master/src/util/Tlv493d_conf.h).
So according to that e.g. the command for setting the `IICAddr` bits to `0b01` would be:
`setRegBits(tlv493d::W_ADDR, 0b01);`

Afterwards the `setAccessMode()` function is called, this will set the bits for lowpower mode, calculate the parity, and write the configuration to the sensor.

electronicDuck, 2022-02-02:
Ok great thank you, I will have a go at making the changes :)

closed 2023-04-05 by jaenrig-ifx