Infineon/TLV493D-A1B6-3DMagnetic-Sensor #23: Cartesian Example only returning X0, Y0, Z0 continuously

giftofyours, 2022-02-19:
Hi Juan,

I seem to be encountering a similar issue as Mark reported 2 years ago in the closed issue:
https://github.com/Infineon/TLV493D-A1B6-3DMagnetic-Sensor/issues/12#issuecomment-492615521

I'm getting readings:
X = 0.00 mT; Y = 0.00 mT; Z = 0.00 mT

Meanwhile, the device works smoothly in the GUI application. The GUI application shows that I have device TLV493D-A1B6.
The board itself is labelled "TLV493D-A1B6_2GO_V3.2"

Tried to use the recommended solution at that time, via branch https://github.com/Infineon/TLV493D-A1B6-3DMagnetic-Sensor/tree/kit2go-wa that you shared on 14 May 2019. However, that branch does not seem to exist any more.

Would appreciate advice how I can get the code working, or how I can edit the library directly to enable that power pin assuming it is a similar issue.

Thanks!
Gi

Anandaramashetty, 2022-03-03:
Hi Gi,

The fix for the issue is already moved to Master branch. You can download latest code from link.
https://github.com/Infineon/TLV493D-A1B6-3DMagnetic-Sensor

You might have forgot to enable these below lines:

https://github.com/Infineon/TLV493D-A1B6-3DMagnetic-Sensor/blob/8945ae676571233a9a974c6e8b0b053e1faf52d7/examples/Cartesian/Cartesian.ino#L11-L13

They are responsible for powering the sensor (the VDD of the sensor is connected to LED2 on the board).

Regards,
Anand

giftofyours, 2022-03-04:
Thanks Anand!

Much appreciated.

closed 2023-04-05 by jaenrig-ifx